uf-mil-electrical/TerraGator #1: Battery Monitor Board

MjAdrian, 2025-02-03:
**Deadline: 04/01/2025**

### Project Description

#### Purpose

The battery monitor board will ensure that the batteries connected to the rover will not be overused. Overusing Li-Po batteries can irreversibly damage them and potentially cause a fire.

#### Desired Behavior

- LED that indicates the voltage is good (22.2-24).
- Send a signal to the computer to indicate is it good or bad 
  - Can either be digital (high for good or low for bad), or analog (0-5V range) where the computer does ADC to estimate the voltage

#### Pre-reqs

- Altium Knowledge
- Digital Logic
- Circuits 1

### Project Requirements

#### Size & Layout

TBD

#### Electrical I/O

- Takes in Battery (6s LiPo Battery)
- Outputs either a digital or analog signal to the computer
- LED indicator (On or Off)

### Optional Features

- LED Matrix that indicate a range

### Relevant Resources

TBD

RuAlMac, 2025-03-20:
03/20 Update: PCB design is done! Ready to order. Everything's been double-checked. I made some design changes last week: moved the LEDs to the top of the board so that they're visible once the board is installed in ReptileRover, also added a second fuse mount so that the board can be run at its rated maximum of 76A continuous / 80A instantaneous.

RuAlMac, 2025-04-07:
04/07: waiting for PCB and parts to be purchased. Wiki article completed.

closed 2025-04-07 by RuAlMac