uf-mil-electrical/TerraGator #2: Motor driver

TheDinner22, 2025-02-03:
**Deadline: 04/01/2025**

### Project Description

We need the rover to move around!!!!

#### Purpose

Software will be able to interact with the DC motors and drive/turn the rover **through** this board.

#### Desired Behavior

- Speed driven by PWM (higher duty cycle, faster motors)
- some kind of digital logic kill system
- different PWN channels for left and right wheels
- each wheel has it's own enable

#### Pre-reqs

- Altium Knowledge
- Digital Logic
- Circuits 1

### Project Requirements

- the rover needs to be able to move!!

#### Size & Layout

TBD

#### Electrical I/O

- Takes in Battery (6s LiPo Battery)
- takes in ~6 different digital signals (enables for each wheel)
- takes in somewhere between 2 and 4 pwm channels from the PI

### Optional Features

- optimized to use only 2 PWM channels instead of 4
- heat monitor system or something with a temperature sensor??

### Relevant Resources

TBD

Shxl2, 2025-02-10:
This week, I spoke with @TheDinner22 about potentially adding encoders to the drive wheels for closed loop velocity control and wheel odometry.

I also discussed the possibility of adding an absolute encoder to the steer axis of the rover, so that it could be fully holonomic. With the way the rover is currently assembled, the steer motor is geared 1:1 so a motor with an integrated absolute encoder could be used, but I haven't been able to find one that's brushed.  An alternative could be to gear an external absolute encoder 1:1 with the steer, so that it could reset from any orientation, and ideally this encoder could also have a quadrature output for more accurate velocity readings. I've found one that fits these specifications, although admittedly it is a bit pricey and we'd need 4. [(https://store.ctr-electronics.com/products/srx-mag-encoder?srsltid=AfmBOoqM6qOF41YZ5dEfK4YTgYAtGN1Aw3j9ZVoLknZhY_iKIv7Uj5SP)](url) 

For steer, the other option that was discussed was a servo, which would be a lot easier to do electrically and mechanically (and a lot cheaper), but the odometry and control would suffer as a result, because there is no good way of telling where the motor is, you just have to believe that it's at the setpoint. 

Regardless of what we do, we need to talk to the mechanical team so they can package the new motors and possibly encoders, and so they can figure out what to do with the wheel in the center, which cannot be on the ground while the rover is moving sideways. The motor driver boards would also need to change. The drive motors can't be controlled with just 2 outputs anymore, and more boards need to be made for the steering actuators.

TheDinner22, 2025-02-28:
leaving open but we have opted to use the following instead (at least for now):

https://www.handsontec.com/dataspecs/module/7A-160W%20motor%20control.pdf