uf-mil-electrical/TerraGator #4: Software + ROS (maybe) + I2C for PI hat

TheDinner22, 2025-02-14:
**Deadline: 04/01/2025**

### Project Description

#### Purpose

PI needs to talk to some hat via I2C. We need software that does that. Ideally somewhat generic since this hat will be driving the DC
wheels and the servos that let the rover turn.

#### Desired Behavior

- only uses 3 pins for I2C
- minimal wiring to make use of the PI hit
- 16 different PWM channels with full use of the 12-bit channel resolution

#### Pre-reqs

- uP
- 3701C
- python
- PI sdk stuff

#### Size & Layout

- ideally you can do this with no extra wiring (excluding the wires to the motor)

#### Electrical I/O

- no extra wires pretty please

#### optional stuff

- maybe put ROS2 two on the PI?!?!?
- maybe a docker environment that mimics the PI's setup.

### Relevant Resources

[here's the hat](https://www.pishop.us/product/servo-driver-hat-for-raspberry-pi-16-channel-12-bit-i2c/)

Shxl2, 2025-02-17:
This week I worked on imaging the pi with Ubuntu 24.01 so that eventually we could put ROS2 Jazzy on it. For now, I'm looking into using docker to ease software development. What I want to do is use a local registry to push and pull docker images, and make a script file that will ssh into the pi and run said docker image. The intent is that anyone can pull the software repository, and with docker installed, can use the IDE of their preference (VSCode, Jetbrains, NeoVim, etc.) to build images, run a script that pulls that image onto the pi, and then run that image on the pi. Because the rover is meant for newer members to work on, I want to make the software development process as simple as possible. To do this, I'm currently learning how to use docker and make local registries. I'm going to talk to some of the software leads about this to see if they have any advice. 

Here's one of the resources I'm looking at: [https://docs.docker.com/get-started/docker-concepts/the-basics/what-is-a-registry/](url)

Because the pi is running Ubuntu (either via docker or with the imaged os) for ROS, we need to use a different GPIO library that on raspberry pi os. The library that is recommended for Ubuntu is LGPIO according to this website: [https://ubuntu.com/tutorials/gpio-on-raspberry-pi#2-installing-gpio](url) This guide goes through how to interface with basic gpio functionalities, i2c, and pwm. I plan on using this library to interface with the pi hat and control all of the actuators.

TheDinner22, 2025-02-23:
### Joe Handsome

PI hat came in this week and I got it working on Ubuntu. All of the code is in the NewCode2025 repo. We can control each of the 16 PWM channel's with functions (and we get to choose the duty cycle). There's also a version for driving servos.

This issue will likely be closed Monday or Tuesday since the code is super duper close to done (lowk will take me 5 min on Monday or Tuesday). 

here's a video of one of the wheels being turned. We should be able to have the whole rover moving around by end of next week.

[image]

TheDinner22, 2025-02-28:
The code worked and the rover moves!! 

Bugs
- turning is inverted (oops)
- code is poorly written (no loops and somehow I'm passing in negative PWM??)
- no graceful shutdown so rover tends to stay running after program is killed
- no ros2 launch so it's sort of involved to spin it up

closed 2025-02-28 by TheDinner22