uf-mil-electrical/TerraGator #5: Servo motor + footprint+ meche

TheDinner22, 2025-02-17:
**Deadline: 04/01/2025**

### Project Description

#### Purpose

We need to pick some kind of 12V or 24V servo motor for the rover. We will need 4 in total.

#### Desired Behavior

Either the servos will work as a drop in replacement for the DC motors that are currently on the rover (same size/mounting mechanism).
OR
Ensure that the servos you pick can be mounted to the rover (pretty please)

### Project Requirements

servos end up on the rover and work to turn the front and back wheels.

#### Size & Layout

TODO

#### Electrical I/O

Try and get 24V servos. 12V servos will require a buck converter.

#### Project Tasks

choose motors
order motors
mount motors

Shxl2, 2025-02-20:
The servo that the mech team is looking into is the Axon mini. [https://axon-robotics.com/products/mini](url) This servo was designed for FIRST Tech Challenge and due to it's design constraints, it has an analog absolute encoder output. We could use this to reset the azimuth upon initialization no matter what orientation it's in. This is a 12V servo so we're gonna have to buck boost down to 12V. The mounting solution is still a WIP.

SzanderB, 2025-03-02:
Quick summary about using Pi Picos as Quadrature Decoders
Purpose: Use a Pico to do quadrature decoding for hopefully 8 motors
Background: [This link](https://forums.raspberrypi.com/viewtopic.php?t=346174) shows a pico handling 8 encoders and then transmitting them via I2C or SPI. Source code is found through the github link in both python and C/C++
Using Programmable Input Output, there are 8 slots in which we can receive encoder data, send to main CPUs, then analyze and send in packets to host Raspberry Pi when needed.
Quick Intro to [PIO](https://www.youtube.com/watch?v=YafifJLNr6I&ab_channel=LifewithDavid)
[YT](https://www.youtube.com/watch?v=jklc2Aq9-1E&ab_channel=XtendedGreg) and [Github](https://gist.github.com/XtendedGreg/83f1c23f43f8bda27ad761195160fbdd) link to a working quadrature decoder in MicroPython. I’d recommend only using this to get one operational. If we need to do 8, micropython might not be good enough.