uf-mil-electrical/TerraGator #6: Pi and driver hats for anti rats nest

TheDinner22, 2025-02-28:
**Deadline: 04/01/2025** (lowk b4 spring break plz)

### Project Description

There is a horrible routing wiring rats nest issue. Most of the boards we are working with rn have male [dupont-like](https://en.wikipedia.org/wiki/Pin_header) connectors

#### Purpose

Standardize connections and make it so that the person wiring up the rover doesn't need to understand where each individual wire goes.

Ideally we will have hats for the pi and the 3 motor drivers.

#### Desired Behavior

Pi hat and motor driver hats should break the respective pins out to [sherlock](https://www.digikey.com/en/product-highlight/m/molex-connector/sherlock-connector-system) connectors.

The sherlock connectors should be labeled ***so that someone connecting them can easily tell which cable goes where***

#### Pre-reqs

- Altium tutorial

### Project Requirements

pins should also be forwarded so that we can stack hats on top of your hat if desired.

#### Size & Layout

Pi and motor drivers are in lab. There are also probably several Pi hat-templates online.

#### Electrical I/O

These are all low voltage signals and your board doesn't need to worry about power, ground, or noise.

### Relevant Resources

I will add the spec here in like 40 minutes

hannahc963, 2025-03-03:
Got assigned to and started working on this task on Friday 2/28. So far I have an Altium project with a PI hat and am working on the PI hat connections to the sherlock connectors.

hannahc963, 2025-03-10:
Have schematic for PI hat done and am working on the PCB layout. Should be able to finish the motor driver hats as well by this Friday.

hannahc963, 2025-03-16:
3/16: Finished PCB layout for PI hat and finished schematic and PCB layout for motor driver hats. Will push two Altium projects into branch pi-driver-hats-2025 by end of today.

hannahc963, 2025-03-31:
3/30: 
Made some final edits:
- Added pours for GND and Power
- Cut Pi Hat in half
- Labels and Logo
- Note: Rules not imported to projects

hannahc963, 2025-04-07:
4/6:
Adrian and Joe looked over my boards and they should be ordered soon.