uf-mil-electrical/TerraGator #7: Sonar with I2C and pico

TheDinner22, 2025-03-01:
**Deadline: 04/01/2025** (b4 spring break plz)

### Project Description

I made a branch for ur code called `sonar_pico_code` (empty rn ofc)

we have these sonar things. Pretty please have the pico constantly ping them and provide the data over I2C when requested. Do the processing on the PICO, so data should be in cm or something similar.

#### Purpose

The use of a pico will free the pi up to do anything else (probably ros stuff).

#### Desired Behavior

PICO abstracts all information about the sonar(s) away and simply provides data over I2C when requested.

#### Pre-reqs

- PICO experience 
- uP
- I2C
- DAD (for understanding how the sonar works maybe ig)

### Project Requirements

#### Size & Layout

Ideally we'd have a small custom pcb for this. However, I think we'll want to modify what the PICO is doing and possibly use it for more in the future. This task is more about the embedded code. ***Just get it working on a breadboard pretty please***.

#### Electrical I/O

4 pins to the sonar (per sonar) 
3 pins to the pi (I2C pin-count independent of the number of sonar).[

### Relevant Resources

pretty sure [this](https://www.digikey.com/en/products/detail/sparkfun-electronics/SEN-24049/22258319?s=N4IgTCBcDaIMoFEByBaMAWADOgnCAugL5A) is the exact part.

JakeWesta, 2025-03-10:
[image]
[image]

Got the sonar working with connecting to the Pico. Sir Handsome and I will work on Monday to take a running average and transmit that to the PI with I2C, should be chill. Heres the wiki page that im dumping some info on. https://milwiki.cbrxyz.com/wiki/HC-SRO4_Sonar

JakeWesta, 2025-03-15:
3/10-3/17:
- In NewCode2025 branch, joe and i worked on making sonar_master, which is a ROS node that takes the sonar code i wrote (revised into python) and publishes it so that it can be listened to. 
- We tried to get I2C working from pico to pi but it was not being nice, so thats why we switched to a ROS node and directly connected to pi.

JakeWesta, 2025-04-07:
3/31-4/7:
- (Technically not the issue I'm on but it doesnt really have a github, besides the point)
- Tested the sonar's detection capabilities mildly extensively for a couple of hours, it CAN work but it is really not optimal, therefore I looked elsewhere and did research on sonar microphones and transducers to see what capabilities are out there that can suit our needs. Going to talk to Sir Lester and Joe about it to see what they think. Awesome.

JakeWesta, 2025-04-14:
4/8-4/13:
- Looked into RobotX description of the challenge (https://robonation.org/app/uploads/sites/2/2024/09/2024-RobotX_Team-Handbook_v1.0.pdf), Page 19) to understand what I would want to model in the rover itself.
- https://micbooster.com/ultrasonic-microphones/146-ultrasonic-mic-board.html shows decent promise, however I am worried with range and the fact that it would require extra post processing and filtering.
- [this one](https://maxbotix.com/products/mb1340?variant=41051366162483&tw_source=google&tw_adid=&tw_campaign=21565434934&utm_term=&utm_source=google&utm_medium=cpc&utm_campaign=AP+%7C+PM+%7C+Tank+Sensors+Canada&hsa_acc=6184366094&hsa_cam=21565434934&hsa_grp=&hsa_ad=&hsa_src=x&hsa_tgt=&hsa_kw=&hsa_mt=&hsa_net=adwords&hsa_ver=3&gad_source=1&gclid=CjwKCAjwzMi_BhACEiwAX4YZUB3XqrOwWeOybboX3fha0evruRfmhHwEhjfSsFvc9OiHeKe0HD4ACRoCyL4QAvD_BwE) is also a solid option.
- ive learned that land based sensor suck with distance so it will be hard to replicate the RobotX challenge well, perhaps scaling down the course would help and be still be valid.
- Had a rough week, but life is good this week and will lock in decisions tomorrow at the meeting.