uf-mil-electrical/TerraGator #10: MAVROS Node

LesterBonilla, 2025-03-06:
# Description
The Rover and Drone will communicate via a protocol called [MAVLink](https://mavlink.io/). We need a node that will connect to the RFD900x radio modem connected to the Pi and continually listen to messages. These messages will be used by the Rover as necessary. We also need a way to send out MAVLink messages. The communication protocol will be developed as the Rover comes online and we decide how to control it.

# Deliverables
Add your code to the relevant folder in the repo. Make sure to comment your code well and create a page on the wiki for it. Keep us up to date with the comments on this issue on your progress.

jesusgil7, 2025-03-31:
I wrote the node that listens from the radio , with mavlink implemented, I will test it this week

jesusgil7, 2025-04-07:
Download ArduPilot to test the nodes , and keep developing the nodes and reading the commands

jesusgil7, 2025-04-21:
made documentation about the ros and the server client