uf-mil-electrical/TerraGator #20: Rover Camera Integration

LesterBonilla, 2025-03-07:
# Description
We have a [camera](https://shop.luxonis.com/products/oak-d-w)! We need to decide how to integrate this into the Rover and get its data. The camera has on-board image processing/AI acceleration that we need to look into. 

# Deliverables
A quick report on what you've learned about the camera
An idea of how you suggest we integrate the camera (I.E. how to implement a vision model on it). We can ask the software team member to work with you on this task.
A ROS node that will stream data from the camera.

SzanderB, 2025-03-13:
https://docs.google.com/document/d/1AMI41eQ0rKKQh_LzfCIfXSjAarU053IaULmf-sOdI7E/edit?usp=sharing

Doc for writing down findings^^ Will transfer over to wiki after were done

nollfrt, 2025-03-14:
Assigning myself to this issue

SzanderB, 2025-04-04:
Week of 3/31: Got camera ros node operational when connected to computer. Code is found in github linked in Wiki Page. Library includes template code for posting raw image data, rgb data, as well as stereo data. Worked on understanding how to use the on board AI acceleration for neural net models for use in recognizing buoys. Minorly edited the Wiki Page. Next steps are editing the Wiki Page to look much better as well as getting a working buoy recognizor model on the camera.

SzanderB, 2025-04-14:
Week of 4/7: Was out for beginning of week and got sick for a few days. Didn't put too much time in but got pictures of buoy and attempted to use MIL computer to make a PyTorch Model for object detection. Computer was broken, so tried doing it on my laptop and could not get label studio to install correctly. Ran out of time, but next week is update wiki and get a model working.

nollfrt, 2025-04-14:
Week of 4/7: Made minor edits to the wiki (added infobox) and began studying and researching how to implement an AI model

SzanderB, 2025-04-20:
Got label-maker to work, labeled buoy image data, got bounding box label info, downloaded ultralytics to train yolo model, trained model, now need to convert to a file type that works with the camera and upload the code

nollfrt, 2025-04-21:
Continued trying to implement AI model. Szander got one to work and was able to train it, so now looking into converting it to a file type compatible with the camera.

SzanderB, 2025-10-28:
After a long semester, I have updated documentation on camera so far. This is all found on the MIL website. I got a custom model trained on it, and it works pretty well. However, the ROS part looks like another crazy undertaking.