uf-mil-electrical/TerraGator #29: [Software] Programming rover!

RuAlMac, 2025-09-10:
**Deadline: 10/31/2025**

### Project Description
We need to write software for the rover. There's lots to do!
- Get Raspberry Pi 5 talking with the Cube Pilot Orange+ sensor hub
- Get Raspberry Pi 5 talking with the RFDx900 radio modem
- Get Raspberry Pi 5 talking with a stereo camera for future machine vision capabilities
- Get controller talking with Pi 5 for manual rover control
- Make it easy to program in future onboard sensors for collision prevention (e.g., several ultrasonic sensors mounted to the outside of the rover to detect if there is a wall approaching)
- Make it easy to program in future sensors for color detection (e.g., a color sensor mounted to the bottom of the rover so that the rover can follow a track on the ground)

#### Purpose
The first priority with this rover is to get it moving at all. That means getting the Pi 5 talking to a controller should be the first task to get done. Past that though, this rover is kind of an open book. We don't have any hard and fast requirements since we're not competing with it, so we can just kind of do whatever we want.

aouress, 2026-02-16:
2/9-2/15: This week I was able to install ROS2 on the rover Pi and began brainstorming the directory architecture for the different packages we will need to write. Hoping to get a keyboard control over ROS working soon.

aouress, 2026-03-29:
3/29: I have been pretty busy but will be tailoring one of my class projects to be a simulation-based version of the rover (or similar) so that I can write some navigation/path planning code for the project and for MIL.

closed 2026-08-19 by RuAlMac