uf-mil-electrical/TerraGator #34: [Electrical] Sensors for collision detection

RuAlMac, 2025-09-10:
**Deadline: 10/31/2025**

### Project Description
Currently, the rover has no way of knowing if it is about to run straight into a wall. In the past, we had an ultrasonic sensor attached to the front of the rover for this purpose. If the rover was approaching a wall, this sensor would detect that and tell the rover to stop or change direction. We want to restore this functionality, but do so more comprehensively.

We want to attach ultrasonic (or some other kind of distance sensor) to each side of the rover (front, back, left, right). These sensors will communicate with the Raspberry Pi 5 to alert the rover of an imminent collision. We also want to make sure that sensors are mounted securely to the chassis, and have a good electrical connection to the Pi. One idea to ensure this is designing a small PCB that a sensor can be mounted to.

#### Definitions
- Ultrasonic sensor: a type of distance sensor. This sensor shoots out a wave of sound at a frequency not audible for humans, and determines distance from an object based on how long it takes for the sound to reflect back.

#### Purpose
Securely mounting sensors to the sides of the rover for collision detection and prevention.

### Project Requirements
- Choose a type of sensor that can be used to detect distance from walls / other objects. This sensor should be small, not super power intensive, and not super expensive.
- Design a PCB that this sensor can be mounted to. Nothing too crazy, just a simple mounting board for our sensors.
- Determine how these sensors will connect to the Raspberry Pi 5 onboard the rover. Will they use a communication scheme (I2C, SPI, etc.)? Do they have an analog output that varies based on distance from an object?

#### Size & Layout
The PCB mount for these sensors should be quite small. Ideally, less than 20x20mm.
Also make sure to put a good connector on this PCB! We don't want to just use header pins / Dupont wires because those do not always provide a good electrical connection. Screw terminals, shrouded headers, Sherlock connectors, etc. are good options.

Once this project is completed, it will be handed off to the mechanical team. They will design a mount for your sensor PCB to be mounted to the chassis.

aouress, 2025-09-24:
Decided to use the VL53L4ED ToF sensor from STMelectronics because it had better mid-to-close range accuracy than the TL53L0X. The first version of the board is complete. Once reviewed, I'll make a wiki page for all the information related to the ToF sensor and breakout board.

aouress, 2025-10-05:
The board is complete and BOM has been uploaded to the repo. Should be good to order whenever we want.

aouress, 2025-10-26:
10/26: The sensors we ordered have a very annoying package for soldering. We ordered a stencil to help with future soldering applications for these LGA packages.  
Found a good level-shifter to convert between the 1.8V and 3.3V logic levels of the ToF sensors and Pico, respectively. The plan is to have a central board that connects all the ToF and ultrasonic sensors to a microcontroller (likely a Pico). The Pico will serve as an "interrupter" to the Pi5 for collision detection purposes.

aouress, 2025-11-10:
11/9: Made a 3.3V -> 1.8V level-shifter board that we can use to breadboard with the ToF breakout board. This board can be connected to one of the ToF breakout boards, or a ToF can be soldered directly onto the board for testing I2C communication and our soldering practices.